Audio-Injector/Ultra #2: Low frequency vibrations - bandwidth from 1-50Hz

tringwebdesign, 2018-07-03:
Will the Ultra be capable of recording vibrations from local traffic [via a seismometer](https://mindsetsonline.co.uk/shop/lego-seismometer-kit/) 

I would be looking to record the voltage changes from the seismometer in the range of 1Hz - 50Hz. I understand the Ultra is capable of ultrasonic recording, but is it capable of "subsonic"?

flatmax, 2018-07-04:
Hi there,

Yes, the base audio card is capable of DC input/output. They are 
differential pairs.

If you look here :

https://github.com/flatmax/AudioInjector.Octo.RCA

You will see schematics and gerbers for the layout of a passive interface.
If you bypass the AC coupling caps, you can go down to DC. If you change 
the default 10uF AC coupling caps to something much bigger, then you can 
lower the cutoff frequency.

Alternatively you can design your own board !

Matt

tringwebdesign, 2018-07-04:
Interesting, but way beyond my skills set and knowledge, I was more after something off-the-shelf

flatmax, 2018-07-04:
Can you solder capacitors onto an existing board ?

tringwebdesign, 2018-07-05:
I have recapped an old Atari 2600 VCS from 1984 and soldered Geteron switches to a 96% mech keyboard PCB, so am comfortable with a soldering iron. I've currently been using the Cirrus Logic Audio Card for this experiment but again can't find out it if the hardware does filtering. [Here's the thread](https://www.element14.com/community/thread/64399/l/is-there-a-low-frequency-cut-off-value)

flatmax, 2018-07-05:
Well my recommendation would be to attach bigger caps in parallel with 
the caps currently on the RCA board - this will lower the AC coupling 
cutoff frequencies.

chobbs22079, 2019-04-26:
I would like to work out how to do this.  So, you are indicating that only the base board is used for the acquisition of the signal?  Of so, the link you pointed to for the octo schematics: if it is the same for the ultra2's baseboard, can you specify which caps in the pdf need to be replaced and how the RCA connectors are mounted on the baseboard?  I could not tell from the pdfs.  Please let me know.  Thank you.

flatmax, 2019-04-27:
The ultra 2 is totally different to the Octo. However the Octo's base board is DC coupled. Basically with the Octo, if you interface its pins directly, you are going straight into the ADC and out from the DAC ! For that reason you may want to use your own preamp board.