Audio-Injector/Ultra #5: Setting up mic for recording

hystrix1, 2018-11-04:
The sound card seems to default to Line In for recording.  This can easily be changed to use the on-board mics by using the following commands:

`amixer cset name='ADC Mux' 0`
`amixer cset name='PGA Volume' 44`

...and record using:

`arecord -r 192000 -f S16_LE -c 2 -d 5 /tmp/test.wav`

A nice ultrasound spectrogram can be found in my post on the audio injector forum: 
http://forum.audioinjector.net/viewtopic.php?f=11&t=5678#p8339

Nice sound card :-)

flyDlight, 2019-02-21:
Thanks for this informations.
I'm trying to build a bat detector, but I can not find an instruction manual for the sound card. Where did you find this commands?

flyDlight, 2019-02-24:
I answer myself. I asked a stupid question. They are the standard ALSA commands, which I did not know.

flatmax, 2019-02-28:
What commands would you like help with ?

I assume you have set up the card ?

alsamixer is an example alsa application for controlling the mixer.

aplay to play arecord to record.

You can also use any of hundreds/thousands of standard Linux audio 
applications.

flatmax, 2019-02-28:
Yes, that is right !