Audio-Injector/Ultra #6: Preamp board Question

MicroMPX, 2018-11-12:
Well my cards have arrived, and now i have a question, since i am only using the LINE inputs and LINE outputs, do i need to use the preamp board ?

MicroMPX, 2018-11-17:
Hi Matt,

Shure, i will use the soundcard for MicroMPX, MicroMPX will transport FM composite over IP.
I need the 192Khz input sample rate since the composite has a 75Khz bandwith.

So normaly the composite is generated by a FM Broadcast processor like Orban or Omnia, it composite output signal will be the input for audio injector ultra.

The soundcard does not fit within the Raspberry footprint, is this correct ? 
I’m planning on using the header connector with the long pinns, but for it to fit within a raspberry footprint it seems i need to use a flat cable, can you confirm this ?

Jeroen Platenkamp
MicroMPX.NL

MicroMPX hardware Powered by PhantomTronics NL 
MicroMPX software Powered by Thimeo audio technology
MicroMPX.nl | Facebook: MicroMPX | MicroMPX = MPX over IP

Van: Matt Flax
Verzonden: dinsdag 13 november 2018 03:58
Aan: Audio-Injector/Ultra
CC: MicroMPX; Author
Onderwerp: Re: [Audio-Injector/Ultra] Preamp board Question (#6)

Hi there,

In general, yes you need to use the preamp board. More specifically, it 
depends on what you are doing with it.

The preamp board for the line input provides AC coupling - which means 
you can have DC voltage differences between the two devices you are 
plugging together. If this is not a problem for you then, you probably 
don't need the preamp.

Can you describe your use case ?

flatmax, 2018-11-18:
> Hi Matt,
>
> Shure, i will use the soundcard for MicroMPX, MicroMPX will transport 
> FM composite over IP.
> I need the 192Khz input sample rate since the composite has a 75Khz 
> bandwith.

>
> So normaly the composite is generated by a FM Broadcast processor like 
> Orban or Omnia, it composite output signal will be the input for audio 
> injector ultra.
>
> The soundcard does not fit within the Raspberry footprint, is this 
> correct ?

It is the size of a Pi Zero, slightly larger on the left side.

> I’m planning on using the header connector with the long pinns, but 
> for it to fit within a raspberry footprint it seems i need to use a 
> flat cable, can you confirm this ?

It is plugable it should be fine. Particularly as you aren't planning to 
use the optical connector, it can sit flush. People who want to use the 
optical connector need to sit the card up a little - as the regular Pi's 
USB connector gets in the way of the optical connector.

flatmax, 2018-11-18:
> Ok Matt,
>
> Thanks for your answere, to bad it does not fit in RPI 3B footprint.
>
> Final question to get it running…
>
> I am using rasbian stretch lite, i have modified the config.txt file,
> and next you state i have to manual bash the folowing strings…(github)
>
> This raises a question (i am a farely newbee) nomal i do this with an
> editor, can you explain what the script name is ?
> Do i need to start it everytime on boot ?
>
> Jeroen Platenkamp
> MicroMPX.NL
>
>
> MicroMPX hardware Powered by PhantomTronics NL
> MicroMPX software Powered by Thimeo audio technology
> MicroMPX.nl | Facebook: MicroMPX | MicroMPX = MPX over IP

The bash script in the github readme is to be run once. All it does is 
updae the pi's firmware to get the latest audio drivers. It then 
disables one line in the config.txt and enables the audio injector ultra 
device tree overlay.

These tasks only need to be run once - so don't put them into an 
autostart script !

Matt