Audio-Injector/Ultra #8: No headers in kit?

guysoft, 2018-11-17:
Hey,
I just got my Ultra 2 packages, and there seem to be no 40-pin header of any sort - how do I connect the card to the Pi?

flatmax, 2018-11-18:
Yes, it doesn't come with 40 pin headers.

What country are you in ? I would imagine your local electornics hobby 
store has 40 pin connectors specifically for the Pi.

You see, we don't know what kind of 40 pin connectors people require. 
Watch the assembly video before you purchase your 40 pin connector.

guysoft, 2018-11-18:
Israel, stuff here is expensive, probably makes sense to order one and wait
a month and save $4 (us). Or I will ask in our communities if anyone has a
spare header.

LyonelB, 2018-12-02:
Hello,
I bought a 40 pins header https://fr.aliexpress.com/item/2-54mm-2x20-Broches-d-tachable-Double-M-le-Embase-M-le-pour-Raspberry-Pi-Zero/32863039215.html

and I welded it well : 
[image]

Then, I fixed the flat ribbon cable
[image]

but it is impossible to assemble the preamp : 
[image]

What should I do?
To undo and solder under the cart?