Audio-Injector/Ultra #10: i want to buy one..or two

dfrhodes, 2018-11-27:
I'm late to the kickstarter party..how do i buy now one of these?

flatmax, 2018-11-28:
We are expecting them to be up on our shop later in Dec. or early Jan. 
https://shop.audioinjector.net/

Don't miss another party, follow us on kickstarter !

undessens, 2018-12-19:
I missed it too ... can't wait to get it ...

flatmax, 2018-12-19:
Thanks for the interest. At this rate, we expect these to be on sale from our shop early January - they are in manufacture as we speak !

richardbeattie, 2018-12-29:
Very interested in getting a few of these also, need them as a critical component of a science project. Any further updates on the timeline?

flatmax, 2018-12-31:
The cards are now live on our shop : shop.audioinjector.net

In a week or two they will be live on amazon US.

undessens, 2019-01-04:
Hi, what about the price ?