Audio-Injector/Ultra #11: Trouble with cs4265 driver and mclk

mario-1024, 2018-12-16:
Hi everybody,

I am trying to get my audioinjector ultra 2 card working, but I am running into this problem:

When booting the raspberry pi 3, I get the following message (dmesg):
[    2.635173] cs4265 1-004e: CS4265 Version 3
[    2.638700] asoc-simple-card soc:sound: cs4265-dai1 <-> 3f203000.i2s mapping ok
[    2.640028] asoc-simple-card soc:sound: control 2:0:0:SPDIF Switch:0 is already present
[    2.642236] cs4265 1-004e: ASoC: failed to add widget SPDIF dapm kcontrol SPDIF Switch: -16

And later repeaded messages:
[    6.849041] cs4265 1-004e: can't get correct mclk
[    6.850080] cs4265 1-004e: ASoC: can't set cs4265-dai1 hw params: -22

alsamixer is showing the card and I can set the volume etc.
When running speaker-test, I get a power supply humming on the speakers, but no sound.

Any hint, where I should be looking for the problem?

Best
Mario

hystrix1, 2018-12-18:
What version of Raspbian are you using? I found that Jessie introduced noise in my ultrasound recordings, and the problem went away when using Stretch. I didn't have time to find out why, so I just make sure I'm using the latest version.

flatmax, 2018-12-18:
Yes, at this point, it seems that the problem may be solved by a driver 
update. rpi-update

mario-1024, 2018-12-18:
I did a rpi-update, which did not seem to change anything, but I did not track the changes in detail.
Then I changed the sample rate to 48000, everything is fine now. When setting the sample rate to 44100, I got the mclk warnings, again.

hystrix1: I use the current zynthian image. And have no sound issues with the current setup.

flatmax: I will try to narrow the 44100 sampling rate issue down next year.

closed 2018-12-18 by mario-1024

flatmax, 2018-12-18:
Oh you didn't mention the sample rate before.
The oscillator on the board is 12.288 MHz. Most sound cards have only one oscillator on board. An exception to this rule is the Audio Injector Octo which has two oscillators.
As the onboard oscillator is 12.288 MHz, it will work natively with sample rates which divide this frequency exactly. Examples ([from here](https://github.com/torvalds/linux/blob/master/sound/soc/codecs/cs4265.c#L246)) are :
32k, 48k, 64k, 96k, 128k, 192k Hz
The other sample rates will require you to use the plug ALSA resampler. These sample rates include :
44.1k, 88.2k 176.4k

Hope that helps.