Audio-Injector/Ultra #12: External Microphone connections

hystrix1, 2018-12-19:
So I don't get it wrong, could someone confirm which pins I need to use to connect an electret mic to, like this one: [https://micbooster.com/primo-microphone-capsules/88-primo-em258n-6-mm-omni-electret-capsules-fc033.html](url)

Also, is it possible to connect an external MEMS mic, like this one [https://micbooster.com/ultrasonic-microphones/146-ultrasonic-mic-board.html](url), and which pins would I connect it to?

Many thanks.

hystrix1, 2018-12-26:
1. Can an external electret mic be connected to the base sound card, and if so, which pins are used and how is the bias voltage applied?  Is any additional circuitry required?

2. Can an external electret mic be connected to the balanced preamp board, and if so, which pins are used and how is the bias voltage applied?  Is any additional circuitry required?

3. Can two electret mics be connected to the sound card, and if so, which pins are used and how is the bias voltage applied to each channel?  Is any additional circuitry required?

Many thanks.

flatmax, 2019-02-21:
[image].

There you can see :

AGND = -

MicBias = +

Microphone1 = out from mic. 1

Microphone2 = out from mic. 2

flatmax, 2019-03-07:
Those connections are most suitable for this :
https://micbooster.com/ultrasonic-microphones/146-ultrasonic-mic-board.html

flyDlight, 2019-03-08:
> Those connections are most suitable for this :
> https://micbooster.com/ultrasonic-microphones/146-ultrasonic-mic-board.html

I do not understand if I can connect it directly or does it need capacitors and / or resistors?

hystrix1, 2019-03-08:
You can connect it directly with no resistors or capacitors - I have just tried it.  The connections from the Ultra main sound card (bottom board) to the ultrasonic mic board are:
MicBias --> +
AGND --> -
Microphone1 --> OUT

flyDlight, 2019-03-08:
Thank you! I like this mic and I will try it.