Audio-Injector/Ultra #17: Only noise can be heard during speaker-test

richardbeattie, 2019-03-15:
Hi,

I've just assembled the board and am only getting noise out of a pair of headphones when I run `speaker-test`. I was for a moment able to hear the correct output but when I tried again after trying from a microphone it had stopped working. 

Steps that were taken to try and resolve:

- Rebooted the Raspberry Pi

- Tried with both a speaker and headphones

It would strike me like a hardware issue although the board is assembled as it is meant to be

richardbeattie, 2019-03-30:
This continues to be an issue. I've been able to get it temporarily working but after enough time it goes back to just giving me noise. It is most likely something to do with the way I assembled the board (I will honestly admit that my experience is limited), however, help to identify ways to solve this would be appreciated.

flatmax, 2019-03-31:
If this is an assembly issue, lets talk offline about this.

richardbeattie, 2019-04-06:
That would be great, how can I contact you?

donalus, 2020-05-24:
@flatmax I think I'm having a similar issue with my Ultra 2. I'd love some ideas on how to debug the hw.

flatmax, 2020-06-25:
We will chat off list.