Audio-Injector/Ultra #21: DC i/o capabilities of the base sound card

DenezThomas, 2019-03-25:
Could you tell us more about the DC capabilities of the base sound card? Especially regarding the pinouts, as we can't get DC from the usual three layer stack of boards.

Thanks!

flatmax, 2019-03-25:
The base codec board can be DC coupled.

The connectors are now up on the wiki here :

https://github.com/Audio-Injector/Ultra/wiki/Base-soundcard

The output lines are buffered, the input lines require you to reference 
the signal ground half rail (SG) to bias your input buffer stage.

You can use the +-12 V rails provided to power op-amps or other chips, 
however don't draw too much current or the supply will fail.

Matt

DenezThomas, 2019-03-27:
Thanks! It works. For now, I'm able to output DC from 0.709v to 3.563v. Ideally, I would scale it to something like 0-10v. 
Is there any way I could modify the existing preamp (something like bypassing AC-coupling caps) or should I design my own op-amp circuit?

flatmax, 2019-03-27:
I think you need to design your own preamp which is DC coupled. The 
preamp we supply is designed to be AC coupled which makes it simpler.

To keep noise down, I recommend using the +-12V we supply.

You can design a very simple preamp using the GND and +12V rail to suit 
your needs. Possibly using a DC removal circuit to allow for swings over 
the entire CV range (+-10V).

Or you can work out how to step the signal ground up from roughly 2v to 
5v and amplify your AC at the same time.

Matt