Audio-Injector/Ultra #26: Matching line level of existing equipment

WesBrenner, 2019-06-12:


WesBrenner, 2019-06-12:
In testing the Ultra 2 as a playback device I encountered what seems to be about 12db higher level than my cd player produces.  This over drives and distorts my equalizer inputs. I solved the immediate problem by the command:
 amixer cset numid=2 91%
however, since this alters the "DAC volume" I assume it happens in the digital domain.  This would essentially only allow 22 bit representation since it takes 2 bits to reduce by 12db.  Rather than giving up these bits, is there a way to change the preamp section to lower it by 12db after the dac in the analogue domain?

flatmax, 2019-08-06:
The peak output levels of the lines are 10V.
One way around this is to use a resistor divider to bring the output voltage levels down to 1V peak.
If you want to do this, you can fab up your own board using this schematic and PCB layout :
https://github.com/Audio-Injector/AudioInjector.ultra.IO

I would use lower value resistors - to keep noise down - as the resistors are in series in a resistor divider, keep the series sum around 1 kOhm which should suit the output line drivers

WesBrenner, 2019-08-09:
Hi Matt:
I was wondering if the opamp on the ADC/DAC board is set up for amplification and whether it is in an inverting configuration.  Seems I might be able to modify it there.  What I'm trying to do is wring out every bit of clean signal from the DAC.  Unfortunately, I see only about 60db difference between a maximum 440hz sine wave and the noise which includes a  left/right clock signal.  These are viewed on an oscilloscope from the 4 pin output connector.  Any suggestions on shielding would be most appreciated.  The wave files are generated with Audacity at maximum and silence 192/32.

flatmax, 2019-08-16:
As a test, what do you see with a 48 kHz sample rate as far as SNR goes ?

The ADC connections for RCA lines are 1:1. The ADC connections for balanced lines is 10:1 gain reduction.
From memory, the DAC has 1:10 gain applied.

You can remove the preamplifier board if you wanted and use an off the shelf preamp. The replacement preamp would require AC coupling because the output from the sound card (before the preamp) is down to DC (DC coupled).

As far as shielding goes, is there no way to have a balanced setup ? It should give the best outcomes.

WesBrenner, 2019-08-17:
Hi Matt:
Thanks for getting back to me..
Actually the tests I did were with only the ADC/DAC board connected to the Rasberry Pi, with no preamplifier, going straight from the 4 pin analogue output to the  AC oscilloscope inputs.
It seems to point to digital noise getting into the analogue output on the ADC/DAC board itself. 
The only solution I can come up with is some how shielding the digital signals to keep them out of the analogue signal.

I will run the tests at 48khz as you requested and report back as soon as I can get to it.
Thanks again
Wes

WesBrenner, 2019-08-17:
Hi Matt:
I have run the tests with a file of silence at 192k, 96k, and 48k.  What is weird to me is that the amount of noise is pretty constant across all three.  Even more weird is that the frequency of the noise is the same in all three  (right at 192k with the left and right channels 180 degrees out of phase).  Any insights as to where this noise is coming from would be most appreciated.

Thanks for the help,
Wes

WesBrenner, 2021-01-29:
HI Matt:

Have you ever had a chance to check this noise source and if so is there a solution?

Thanks for any help,
Wes