Audio-Injector/Ultra #28: Pinout For RPi Connector On Audio Injector Ultra

rahealy, 2019-08-04:
Greetings,

May I please have a listing of the pins and their functions on the RPi 40 pin header that are used by the Audio Injector Ultra?

I'm doing baremetal OS-less development on an RPi3. At this time it would be more convenient if I could solder in hookup wires instead of sourcing a 40 pin header.

Appreciatively,

-Richard

rahealy, 2019-08-04:
Please disregard. After enabling javascript on my web browser for kickstarter.com I found the details here:

https://www.kickstarter.com/projects/1250664710/audio-injector-ultra-2-sound-card/faqs

> Which GPIO pins are used by the Ultra 2 ?
> 
> The shared pins are :
> 5v power : 2, 4
> I2C (shared) : 3, 5
> GND : 6, 9, 14, 17, 20, 25, 30, 34, 39
> 3.3v power : 1, 17
> 
> The reserved pins are :
> I2S : 12, 35, 38, 40
> Reset : 29

Closing issue now. Thank you for your work.

-Richard

closed 2019-08-04 by rahealy

flatmax, 2019-08-06:
Here are the pins used by the Ultra :

2,4 : 5v

1 : 3.3 V

3,5 : I2C

9, 6, 14, 20, 25, 30, 34, 39 : GND

29 : reset

12, 35, 38, 40 : I2S

rahealy, 2019-09-06:
In case anyone comes across this from a search here's the pinout of a ribbon cable that seems to work with a RPi3

```
Ribbon Cable:
Wire Pin Desc
--------------
1    2   (5.0v)
2    1   (3.3v)
3    3   (I2C)
4    6   (GND)
5    5   (I2C)
6    9   (GND)
7    12  (I2S)
8    17  (3.3v)
9    29  (Reset)
10   30  (GND)
11   35  (I2S)
12   34  (GND)
13   38  (I2S)
14   39  (GND)
15   40  (I2S)
```