Audio-Injector/Ultra #29: Should arecord -l detect the presence of the ultra 2 sound card

chobbs22079, 2019-09-06:
I built the sound card.  I edited the config.txt file.  I loaded the latest NOOB and updated after bootup. There was a dtoverlay device listed already so I just added another:

dtoverlay=vc4-fkms-v3d
max_framebuffer=2
dtoverlay=audioinjector-ultra

typing arecord -l gives 'no soundcards found'

do I need to do anything different to test if the ultra is recognized by the system?

Please let me know.  Thank you.

chobbs22079, 2019-09-06:
nevermind.  That section of the config.txt file was in the [pi4] section.  Moving dtoverlay=audioinjector-ultra just above the [pi4]line and rebooting made raspbian recoginize the sound card.  sorry to bother.

chobbs22079, 2019-09-06:
Spoke too soon.  The system can now see the card, but I do not know how to use it. 

What options should work with the arecord program?

I tried:

arecord -D hw:0,0 -d 5 -f cd test.wav 
since the card is listed as card 0, device 0

The error is 'arecord set_params:1403: Unable to install hw params'

Any suggestions welcome on how to record and play back anything with this built in program (arecord).  Thank you.
-- | --

chobbs22079, 2019-09-06:
please let me know.  thank you.

flatmax2, 2019-09-08:
It seems that the wrong hardware parameters are being used.

Can you try first to use the plughw device :
arecord -D plughw:0 -d 5 -f cd test.wav

If this works, then get arecord to print out the hardware parameters you are using when not recording through plughw :
arecord -v -D hw:0 -d 5 -f cd /tmp/test.wav

If it is a 44100 Hz type of sample rate, that is most likely the problem. If that was the problem then try to set a suitable hardware sample rate :
arecord -v -D hw:0 -r 48000 -d 5 -f cd /tmp/test.wav

paddygoat, 2019-09-14:
1. Now click the pi icon (top left).
2. Preferences.
3. Audio device settings.
4. Set controls.
5. Tick the first 9 boxes.
6. Raise DAC levels to top.
Now works !!!!