Audio-Injector/Ultra #31: Bare Metal RPi3 With Ultra 2

rahealy, 2019-11-07:
Greetings,

I'm currently working on a "Raspberry Pi 3 Bare Metal Effects Processor" using an Ultra2 sound card:

https://github.com/rahealy/rpi3fxproc

I'm having difficulty getting any sound out of the card. The card works fine in Raspbian so I'm pretty confident that there are no hardware issues.

The first assumption I'm making is that the card is acting as the I2S master (RPi PCM peripheral is slave) and is deriving its clock from an onboard 12.288 MHz oscillator feeding MCLK of the CS4265. Is this correct?

Thank You,

-Richard

flatmax, 2019-11-07:
Hi,

Sounds like a great project!

Yes your assumptions are correct. 12.288 MHz low jitter master clock 
into the cs4265 codec. The Codec is I2S master.

If you run into trouble along the way, consider capturing the state of 
the codec registers. You can hack the Linux cs4265 driver to print its 
own registers out when changes such as starting and stopping the 
capture/playback occur. You could then do a copmarison to the state of 
the registers at the same points in time on your bare metal system.

Regarding the Pi's audio output, are you planning to use DMA ? You may 
be able to print the state of the I2S registers on the Pi using Raspbian 
to also validate your approach with a known working setup.

rahealy, 2019-11-09:
Thank you! Got some sound out!

> Regarding the Pi's audio output, are you planning to use DMA ? 

The plan is to get things running with polling, get a benchmark then determine whether to pursue DMA from there. DMA would be fun to learn and implement but not a requirement at this time.

> You may be able to print the state of the I2S registers on the Pi using Raspbian to also validate your approach with a known working setup.

It turns out that I made two mistakes while configuring the RPi3's PCM peripheral for I2S operation. The first mistake was missing that the frame length is zero based. I set it to 64 instead of 63 (32 * 2 - 1). The second mistake was setting the channel widths to 24 bits (sample rate) instead of 32 bits (actual channel width for I2S).

For the sake of sharing knowledge here's how I went about getting the known good I2C and I2S register contents in Raspbian. 

### I2C
In Raspbian on the RPi I used the i2c command line tools.

Detect the common name of the I2C bus ("bcm2835 I2C adapter"):
```
>$ i2cdetect -l
i2c-1	i2c       	bcm2835 I2C adapter             	I2C adapter
```

According to the CS4265 datasheet the I2C bus address is pin selectable for either 0x4E or 0x4F. Determine which the Ultra2 uses (0x4E):
```
>$ i2cdetect -y "bcm2835 I2C adapter" 0x4E 0x4F
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
40:                                           UU --
```

Dump registers 0x01 to 0x12 while playing a wav file in Audacity:
```
$ i2cdump -f -y -r 0x01-0x12 "bcm2835 I2C adapter" 0x4E b
*snip*
```

### I2S
In Raspbian on the RPi I used an edited version of the devmem2 command line tool:

https://github.com/rahealy/tools/blob/master/devmem2.c

Dump the contents of the I2S registers while playing a wav file in Audacity. In Raspian the 32bit I2S registers are accessed at memory addresses 0x3F203000-0x3F203020:

```
>$ sudo ./devmem2 0x3F203000 ; sudo ./devmem2 0x3F203004 ; sudo ./devmem2 0x3F203008 ; sudo ./devmem2 0x3F20300C ; sudo ./devmem2 0x3F203010 ; sudo ./devmem2 0x3F203014 ; sudo ./devmem2 0x3F203018 ;  sudo ./devmem2 0x3F20301C ; sudo ./devmem2 0x3F203020
*snip*
```

closed 2019-11-10 by rahealy