Audio-Injector/Ultra #33: cs4265 1-004e: CS4265 Device ID (10). Expected 1

LouDou, 2020-06-25:
Hi,

I have attached the Ultra 2 to my rpi 4 and set up the /boot/config.txt as documented.

Device info:
Model           : Raspberry Pi 4 Model B Rev 1.2

$ uname -a                                        
Linux gp-pi 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux

In the dmesg log after boot I have this message:

`cs4265 1-004e: CS4265 Device ID (10). Expected 1`

and when the machine is up, it has not detected the sound card at all.

Do you have any clue what's happening here? Can I fix it in hardware or software configuration?

Regards,
Doug.

flatmax, 2020-06-25:
Oh, lets see if we can do something about this.
Can you send an image of the card attached to the pi, just to check the polarity is right first.

LouDou, 2020-06-25:
[image]

LouDou, 2020-06-25:
Hey, sorry for the bother - I have re-seated all the connectors and now I have success :)
Not sure what happened with the inital setup;

kernel logs
```
[    4.345588] cs4265 1-004e: CS4265 Version 3
[    4.352817] asoc-simple-card soc:sound: cs4265-dai1 <-> fe203000.i2s mapping ok
[    4.353076] cs4265 1-004e: ASoC: failed to add widget SPDIF dapm kcontrol SPDIF Switch: -16
```

alsamixer device list
```
┌───── Sound Card ──────┐
│-  (default)           │
│0  audioinjector-ultra │
│   enter device name...│
└───────────────────────┘
```

closed 2020-06-25 by LouDou

flatmax, 2020-06-25:
Good news. You're up and running !