Audio-Injector/Ultra #34: No I/O -- Assembly Issue?

jatimo, 2020-09-16:
Got my first Ultra 2 board a week or so ago and assembled. 

Device shows up with "arecord -L" list and mixer controls work. But when I attempt to record or play, no data goes in or out. 
The ultra 2 is the only sound device currently active on the unit, so the below are defaulting to this device (ALSA device listing included later).
```
root@VIR32530:/opt/atom# arecord -f S16_LE -r 48000 -c 2 -d 3 test.wav
Recording WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
arecord: pcm_read:2145: read error: Input/output error
```
About an 8-10s pause before the error line is output. With 1 channel requested it hangs forever until I Ctrl-C.
```
root@VIR32530:/opt/atom# arecord -f S16_LE -r 48000 -c 1 -d 3 test.wav
Recording WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
^CAborted by signal Interrupt...
arecord: pcm_read:2145: read error: Interrupted system call
```
In both cases a 44 byte file is produced (just the RIFF header it seems, no actual data). 
I tried changing the ADC Mux and Channel Select and other mixer settings that seemed they might affect what was being captured, but nothing seemed to have an effect. I likewise tried playing sound  with similar results. No sound output, command hangs forever until Ctrl-C interrupt.

```
root@VIR32530:/opt/atom# aplay file_example_WAV_2MG.wav 
Playing WAVE 'file_example_WAV_2MG.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Stereo
  ^CAborted by signal Interrupt...
aplay: pcm_write:2053: write error: Interrupted system call
```
I got a second ultra 2 unit to try to rule out assembly errors. Both obtained by Amazon US, v1.23 boards. Had no issues with assembly and it all looks good, but this one actually does not do any I/O after the initial driver showing up. The same i/O hang above will occur, but this one also cannot use alsamixer. Changing any mixer value populate the field with a 0 value instead of the input amount, and trying to restart alsamixer later gives the below error. This same error seems to come up after any attempt to communicate with the card. dmesg looks normal, just the usual  I2S sync message which is said to not be an issue.  

I've double checked the solder connections under a magnifier and all looks good, but thought I'd check here first before I start individually probing all the connections to be sure I haven't messed up the assembly in a way that's not visually apparent, or on the off chance I'm just the first to try with the newest kernel and there's an issue there. I did perform the updates and /boot/config.txt changes per the instruction page.

`
root@VIR32530:~# alsamixer
cannot load mixer controls: Remote I/O error
`

 System info below, thanks in advance for any input:
```
root@VIR32530:/opt/atom# arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
jack
    JACK Audio Connection Kit
pulse
    PulseAudio Sound Server
default:CARD=audioinjectorul
    audioinjector-ultra, bcm2835-i2s-cs4265-dai1 cs4265-dai1-0
    Default Audio Device
sysdefault:CARD=audioinjectorul
    audioinjector-ultra, bcm2835-i2s-cs4265-dai1 cs4265-dai1-0
    Default Audio Device
dmix:CARD=audioinjectorul,DEV=0
    audioinjector-ultra, bcm2835-i2s-cs4265-dai1 cs4265-dai1-0
    Direct sample mixing device
dsnoop:CARD=audioinjectorul,DEV=0
    audioinjector-ultra, bcm2835-i2s-cs4265-dai1 cs4265-dai1-0
    Direct sample snooping device
hw:CARD=audioinjectorul,DEV=0
    audioinjector-ultra, bcm2835-i2s-cs4265-dai1 cs4265-dai1-0
    Direct hardware device without any conversions
plughw:CARD=audioinjectorul,DEV=0
    audioinjector-ultra, bcm2835-i2s-cs4265-dai1 cs4265-dai1-0
    Hardware device with all software conversions
usbstream:CARD=audioinjectorul
    audioinjector-ultra
    USB Stream Output

root@VIR32530:/opt/atom# lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

root@VIR32530:/opt/atom# uname -r
5.4.65-v7l+

root@VIR32530:/opt/atom# cat /boot/config.txt
#########################
avoid_pwm_pll=1
start_x=1
gpu_mem=256
dtparam=i2c_arm=on

# enable the AudioInjector.net sound card
dtoverlay=audioinjector-ultra

root@VIR32530:/opt/atom# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- UU -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --     

```

flatmax, 2020-09-16:
ok, lets check a few things.

First of all how are you plugging the sound card into the Pi ? Lets make 
sure it is oriented the correct way.

Are you testing on stock Rasbian ? If not it may be better to test on 
stock Raspbian - however may not be necessary.

Before you try to record, probably best to validate the i2c connection. 
Alsamixer does this, however it may have problems with the latest Linux 
kernel - I notice you have a 5.4.xx version which is very recent.

You can try to roll back the kernel for testing purposes - the following 
will roll it back to 4.19.118 from a few months ago :

sudo rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2

I assume that some of the i2c functionality is working because the Linux 
kernel is detecting the soundcard.

Finally, I am not sure that the sound card can read 16 bits natively, it 
is possible it can. You can either change S16_LE in your record scripts 
to S32_LE or you can change the recoding device to plughw:0 which is the 
software plugin version of hw:0 :

arecord -f S16_LE -r 48000 -c 2 -d 3 -D plughw:0 test.wav

or

arecord -f S32_LE -r 48000 -c 2 -d 3 test.wav

Matt

jatimo, 2020-09-17:
Thanks so much for the prompt reply and guidance.

Using a stock RasPi OS did work. All other combinations did not (including rolling back the kernel to 4.19.118). We have a few I2C devices on our system (though none attached at time of testing Ultra2) so something about their configurations must've been in conflict. When I figure out what it is I'll post my findings in case someone experiences the same issue.

Thanks again!

jatimo, 2020-09-18:
The conflict was with physical pin 29, GPIO5, which according to the posts I found regarding pinout for Ultra 2, is a "Reset" pin. Our system is using the x728 UPS HAT to provide uninterrupted battery power, and this board also used GPIO5 (as the power button), resulting in the pin appearing to be driven low from the Ultra 2's point of view.

flatmax, 2020-09-18:
Yes, that is a problem, it completely resets the sound card. x728 UPS 
HAT's gpio5 pin, what is it used for ?

jatimo, 2020-09-18:
It is the power/reset button for the UPS, Its normally  high and pushing the button take the pin low, which is not desirable if you wish to just stack the Ultra2 on top of it. 

For our application, we have a another board that the 40 pin connector is passing through, so there we can just redirect the x728's GPIO5 to another pin, and change the pin value in the scripts that implement the soft reboot/halt. That seems to be working fine for us. We are all good to go.

I was wondering though -- can pin29 on the ultra2 be simply driven high for extended operation, or does the driver or some other mechanism use it dynamically to interact with the sound card at some point, such that it needs to stay connected specifically to that pin of the RasPi?

That's what I did while investigating and it worked OK (clipped the bottom to isolate it, jumpered to GPIO6, then drove that high), but those were short tests  only to confirm that pin was at the heart of the issue.