Audio-Injector/Ultra #36: ULTRA alsamixer mut

mheidt, 2022-05-10:
@flatmax 
After years I try to use the Ultra again. But I have some issues with the latest zynthian/raspbian build.
Generally it works. But the DAC slider is muted after each reboot.
I am not quite sure, if zynthian is doing something during startup, but I will figure out.
BUT.
amixer -c 4 sset 'DAC' unmute

mutes the mixer and

amixer -c 4 sset 'DAC' unmute
mutes it.
Maybe zynthian or raspbian sends an unmute and because your driver is wrong, it mutes it?

flatmax, 2022-05-10:
If you run ( with any necessary args) :
`alsactl store`

Does it store the state of the mixer and apply it correctly at reboot ?

mheidt, 2022-05-11:
no, i was trying that as well.

mheidt, 2022-05-12:
although...what are the necessary args?

flatmax, 2022-05-13:
Good point !
I think the last argument should be the card number or id. I think you can find text IDs from aplay -L
If the ultra device id never changes, then you could use its number.
If the ultra is not the only card on the system and not default, then it is likely that alsactl is not storing and restoring the state of the ultra.

If you are having problems where pulse is taking default's place then you may be interested in this :
https://github.com/alsa-project/alsa-lib/issues/134

I have found that adding this to the top of .asounrc overrides pulse taking default :
```
@hooks [
        {
                func load
                files [
                        "~/.asoundrc"
                ]
                errors false
        }
]

```

mheidt, 2022-05-13:
@flatmax,
could you link me to the source code of the driver? I would like to find the bug myself.

mheidt, 2022-05-13:
when I alsactl monitor and let the zynthian do some volume change:
alsactl monitor 
node hw:4, #21 (2,0,0,DAC Switch,0) VALUE

And it's mute again...

mheidt, 2022-05-13:
How do you explain, that mute is unmuting and unmute is muting?

flatmax, 2022-05-15:
This is strange behaviour. Does this always happen ?

The ultra doesn't have a machine driver it uses the linux simple audio card machine driver, which is setup in the device tree overlay :
https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm/boot/dts/overlays/audioinjector-ultra-overlay.dts

If you make changes, recompile that overlay and reboot then it should allow you to modify it.

The cs4265 codec has a mute function in its driver, you can see that here : https://github.com/raspberrypi/linux/blob/rpi-5.15.y/sound/soc/codecs/cs4265.c#L380
It seems very straight forward and from looking at the mute function I can't see why it would be failing.

When I look at the circuit, it seems like it should operate as expected. The !mute pin on the codec output is low when we are in mute and goes high when we are out of mute. The Preamplifier electronics are setup to operate on that signal. A measurement of the voltage on the mute

I've added a template for the pinouts of the ultra 2 soundcard. You can see the mute pin here : https://github.com/Audio-Injector/Ultra.preamp.template/blob/main/AI.ultra.preamp.template.sch.pdf
It is the bottom right pin. You can reference the Pi's GND plane or one of the AGND pins.

mheidt, 2022-05-17:
Yes, that's weird.
Maybe it's a bug in amixer.
When I run alsamixer, it's mute, when the mm turns into a 00. 
Does this prove, that amixer is the source of the issue?

flatmax, 2022-05-17:
I guess so.

Would it suffice to be able to get the mute to enable at boot time and not operate in zynthian ? I guess you tried this with the alsactl store and restore options.

If it were possible to alter the mute functionality in the device-tree I could help you to generate a new device-tree overlay to do this.

mheidt, 2022-05-21:
The call is also called when I enter a certain screen. 
For now I change the python script.
But I have to do it everytime I update the software.
That device-tree option sounds promissing. Does it mean, that I can make the mute an unmute and vice versa?

flatmax, 2022-05-22:
I guess this mute inversion would have to be part of the ALSA soc simple-card machine driver. If it were possible to override the mute code, then yes.

mheidt, 2022-05-24:
If it's not like "Add this file here" but needs compiling, we can close this.

Thanks a lot Matt

closed 2022-06-03 by mheidt